LibrePCB-Libraries/LibrePCB_Base.lplib #6: Contradicting footprints by different manufacturers

HBadertscher, 2018-11-07:
Hi, 

I am creating footprints for the SO-8 package, but even though the package is standardised by JEDEC, there are multiple contradicting footprint recommendations by different manufacturers:

 - [Fairchild](https://www.fairchildsemi.com/package-drawings/M0/M08A.pdf): 1.75 x 0.65 mm
 - [Centralsemi](https://www.centralsemi.com/PDFS/CASE/SOIC-8PD.PDF): 1.78 x 0.72 mm
 - [Intersil](https://www.intersil.com/content/dam/intersil/documents/m8_1/m8.15.pdf): 2.20 x 0.6 mm
 - [Diodes](https://www.diodes.com/assets/Package-Files/SO-8.pdf): 1.505 x 0.802 mm

How should I handle this? Should we create multiple variants of the package? Or just settle for some reasonable default?

ubruhin, 2018-11-07:
Hi @HBadertscher :)

> I am creating footprints for the SO-8 package

FYI, there is a generator for SOIC packages in development: https://github.com/LibrePCB/librepcb-parts-generator/pull/17

> but even though the package is standardised by JEDEC, there are multiple contradicting footprint recommendations by different manufacturers.

Yes, that's indeed very annoying :(

> How should I handle this? Should we create multiple variants of the package? Or just settle for some reasonable default?

In my opinion it doesn't make sense to have many different footprints of exactly the same package just to follow all manufacturer recommendations. I would rather go with the pad sizes etc. recommended by IPC7351, also because it has the concept of density levels (which most manufacturer recommendations don't have). So basically there should be only one SO-8 package, which should contain three footprint variants for density levels A, B and C.

dbrgn, 2018-11-08:
Since we already have multiple packages with footprints called "hand soldering" and "reflow", we could map the three density levels to "hand soldering", "reflow" and "compact".

The latter is tricky, because it's quite dependent on manufacturer specific tolerances, since you want the smallest possible footprint (which won't work equally well for all manufacturers using the same footprint).

dbrgn, 2018-11-09:
PR: #8.

@HBadertscher does that footprint work for you? There are both EIAJ and JEDEC variants.