LibrePCB-Libraries/LibrePCB_Base.lplib #29: Add transistor devices.

BlueManedHawk, 2019-05-06:
There are currently no transistor devices in the library.  Could these please be added?

dbrgn, 2019-05-07:
Do you have any concrete parts that you'd want in the library?

BlueManedHawk, 2019-05-07:
I was thinking bipolar transistor devices in the SOT-23-3 format.  I've created the models for these in a local library, but I don't know how to add them.

dbrgn, 2019-05-07:
Cool! We generally use manufacturer-specific libraries, so you'd create a pull request against the [corresponding library](https://github.com/LibrePCB-Libraries). Are you familiar with git and GitHub? (If not, and if you don't have the motivation to learn those things, you could also send us the library element through e-mail and we'll add it.)

If the library for the manufacturer of your library does not yet exist, let me know and we can create one.

You can find our library conventions [here](https://docs.librepcb.org/#libraryconventions).

BlueManedHawk, 2019-05-07:
I skimmed the library conventions, and I feel as though a bipolar transistor in the SOT-23-3 format would be a generic component.  Should I do a pull request on this library?

dbrgn, 2019-05-08:
I think you can reuse the generic `Transistor NPN` and `Transistor PNP` components in the "LibrePCB Base" library, right? (cc @ubruhin)

BlueManedHawk, 2019-05-09:
Yes, but I'm wondering whether I should do a pull request to add the appropriate devices.

dbrgn, 2019-05-09:
Yes, definitely! Again, what's the manufacturer of those transistor devices? The `LibrePCB Base` library is only meant for generic parts, so if for example you have a transistor device from Diodes Incorporated, you'd add it to https://github.com/LibrePCB-Libraries/Diodes_Incorporated.lplib.

In order to easily create a pull request, you can clone the git repository to your local libraries (usually `LibrePCB-Workspace/v0.1/libraries/local/` and remove the remote library from `LibrePCB-Workspace/v0.1/libraries/remote/`.

(We should really create a guide for this process...)

ubruhin, 2025-10-04:
Yeah I think we can consider those SOT23 transistors as generic enough to be added to the base library 👍 

> If needed I can do a PR myself to help.

Sure, I'd appreciate any help! The research which transistor types and which MPNs should be added can be quite time consuming though. I remember from #126 I was reading through datasheets, Wikipedia and Digikey several hours. The MPNs listed in the datasheets often did not correspond exactly to the MPNs available on Digikey so it was not straight forward which MPNs to be added to the devices. If you like, I could help with the research & adding the MPNs after you added the devices.

eto-, 2025-10-04:
I started working on PR #161
If the devices are fine I will add to the same PR other transistors in the next days.