LibrePCB-Libraries/LibrePCB_Base.lplib #61: 7400 Parts

ouabache, 2020-03-30:
I have about 20 7400 parts ready to go in a library.

You may want to create a new library for digital parts since there will eventually be a lot of them.

Which library should I add them to?

John Eaton

dbrgn, 2020-03-30:
@ubruhin did you want to create a "logic" library, or should they go into a base library?

@ouabache I wanted to take a look at your open PRs, but so far haven't found the time to do so, sorry.

ouabache, 2020-03-30:
I am worried that if we put everything into the base that it will grow to be unmanageable. Splitting off connectors was a good idea and I think that digital is big enough that it could use its own library. We can use the IC library for analog parts and create a Logic lib for digital.

I have a complete set of symbols for the first 20 7400 parts. These include full chip plus gates plus demorgan gates. I also have 2 different sizes of gates. If you could create a LibrePCB_Logic,lplib then I will check them in.

I did follow the industry practice of putting power and ground on the top and bottom. This violates your guidelines but I feel that it will be better accepted by users. Let me know if I should change this.

I will also create a set of generic devices for both thru hole and surface mount parts.

dbrgn, 2020-03-30:
> I did follow the industry practice of putting power and ground on the top and bottom. This violates your guidelines

This is more of a rough guideline (e.g. for ICs), I think for well-established symbols it might make sense to put them on the top and bottom.

I also think creating a separate `LibrePCB_Logic` library for digital logic symbols / components might make sense. The concrete devices would then go into manufacturer-specific libs I guess. Do you agree @ubruhin?

ubruhin, 2020-03-30:
I would say [LibrePCB_ICs.lplib](https://github.com/LibrePCB-Libraries/LibrePCB_ICs.lplib) is fine for the manufacturer independent elements (mainly symbols+components).

> The concrete devices would then go into manufacturer-specific libs I guess.

:+1:

ouabache, 2020-03-30:
OK, I will add them them into that repo

ouabache, 2020-03-30:
Everything is checked in

ouabache, 2020-03-30:
OK, So two of the symbols failed with a "non-canonical file" error. How can I debug this?

dbrgn, 2020-03-31:
> OK, So two of the symbols failed with a "non-canonical file" error. How can I debug this?

You can open the symbols in LibrePCB, append an "x" (or any letter) to the name, then save the part and run "git diff" in the library. Then you'll see how the generated part *should* look like.

ouabache, 2020-03-31:
Thanks, That worked.

I had been editing the uuid_cache files so that my generated files would
match my originals and some spaces slipped in. That did it.

ubruhin, 2023-09-11:
As discussed, this issue is not related to the base library so I'll close it.

closed 2023-09-11 by ubruhin