LibrePCB-Libraries/LibrePCB_Base.lplib #65: cmpcat for manufacturers

ouabache, 2020-04-12:
We need a cmpcat named "Manufacturers" created at the root level. This will allow any library to create cmpcats for their manufacturers and they will have a place to host it. Then users can search for parts either by function or by manufacturer.

We also need a cmpcat for a generic manufacturer to serve as a placeholder for common parts with no particular data sheet. I suggest using "Generic" or  "Acme Inc"

dbrgn, 2020-04-13:
I'm not convinced that components should be organized by manufacturers. Isn't that what tags/keywords are for?

ouabache, 2020-04-13:
Thats easy if you are looking for something from STM but do I need to know
how to spell Isabellenhutte?

ubruhin, 2020-04-13:
I'll close this issue since there is already a [discussion in the forum](https://librepcb.discourse.group/t/cmpcats-for-manufacturers/219) and we should first decide there how to handle manufacturers.

closed 2020-04-13 by ubruhin