LibrePCB-Libraries/LibrePCB_Base.lplib #68: no device for the selected components

kherwata, 2020-04-17:
hello, i wish you have a good day :) 

I was looking for a footprint of  transistor npn bd137 in this website : https://www.snapeda.com/parts/BD137/10/ON%20Semiconductor/view-part/#download-modal

when i click on download .. it  shows a list of formats for various PCB softwares ... but no format for LibrePCB so i don't know how to add it ..thanks for this software

dbrgn, 2020-04-17:
SnapEDA does not support LibrePCB, so you will have to draw the part yourself. LibrePCB already includes symbols/components for "Transistor NPN", so you would only have to create a device and assign the appropriate package. See https://docs.librepcb.org/#gettingstarted.

> and i want to ask you .. How to Use a Multimeter to Measure Voltage :)

Check out YouTube or use your favorite search engine to answer this question. There are a *lot* of resources for electronics beginners on the internet :)

dbrgn, 2020-04-17:
Note: The TO-126 package (footprint) has not yet been added to the standard library so far. I converted your question to a package request. It might take a while until someone provides the required footprint, so you may want to draw it yourself. Check out the "getting started" documentation linked in my previous comment.

kherwata, 2020-04-18:
Okey .. i want to ask you if i can do the simulation ?

dbrgn, 2020-04-20:
@kherwata no need for closing this issue, having a TO-126 package in the library would be useful :slightly_smiling_face:

ubruhin, 2023-09-11:
TO-126 was added in #95.

closed 2023-09-11 by ubruhin