LibrePCB-Libraries/LibrePCB_Base.lplib #88: Inconcistancy in notation of package size

doom-fr, 2021-02-05:
The capacitors are in imperial units and the resistors in mm

[image]

Same notation should be used.

dbrgn, 2021-02-05:
Yep, the resistors are already using IPC conventions, the capacitors not yet.

ii8, 2023-08-12:
Is it really just the names that are wrong? I initially used a C-1005 capacitor and saw that it was roughly 1mm in length (1.113mm for some reason) so I thought they were all metric. But now I see some of my design looks a bit odd. The C-0603 for example is 1.668mm(0.06569") long and C-2012 is 2.198mm, it all seems a bit all over the place. I'll have to re-check all the capacitors.

To me this looks like quite a severe problem that can result in completely broken designs.

I'd like to fix this ASAP but I don't have access to the IPC standard, is there any way somebody could share the relevant sections with me?

It's also worth it to write a script for these so they are consistent, for example C-1808 doesn't have the glue like the others.

@ubruhin could I get your input, I hope I'm not being an idiot and misunderstanding something.

ubruhin, 2023-08-12:
@ii8 you're right that the capacitor packages are unfortunately quite inconsistent. As far as I remember, these are some of the very first LibrePCB packages ever created and I think they were originally imported from an Eagle library. I also noticed that their dimensions are a bit unusual (not wrong, but larger than needed).

I totally agree they should be improved in several aspects (dimensions, density levels, IPC naming, ...), ideally with our parts generator. Actually this is on my personal ToDo list but I'd highly appreciate if you are willing to take over this task :slightly_smiling_face: 

Relevant documents are these:

- https://www.pcblibraries.com/downloads/Guidelines!IPC-7351B-Land-Pattern-Naming-Convention.asp
- https://web.archive.org/web/20190824094512/http://pcbget.ru/Files/Standarts/IPC_7351.pdf
- https://web.archive.org/web/20190712122301/http://www.ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf

At the moment I don't know if there's a standard for chip capacitors. For the chip resistors there was none, thus @dbrgn did lots of research with the datasheets of several large resistor manufacturers to determine reasonable dimensions, see the discussion [here](https://github.com/LibrePCB/librepcb-parts-generator/pull/29). Unfortunately this is quite time consuming, but I think it leads to really great packages as a result.

ubruhin, 2023-09-06:
Fixed in #107. Some of the capacitors with legacy naming are still there but are now marked as deprecated.