LibrePCB-Libraries/LibrePCB_Base.lplib #103: LQFP package pads narrower than leads

ii8, 2023-08-12:
For some LQFP packages like LQFP50P1200X1200X140-64 for example the pad width is 0.2mm but the lead width is 0.22mm. Is that really what the IPC standard says? Most manufacturers recommend 0.3mm width pads.

ii8, 2023-08-12:
I guess it does say that:
[image]
But in that case shouldn't Density level A pads be wider than the leads, they are currently the same width.

ubruhin, 2023-08-15:
Without taking a look at the relevant generator code yet, I agree this sounds wrong!

ubruhin, 2023-09-15:
Hm it seems this is actually indeed what IPC recommends, respectively the Library Expert: https://www.pcblibraries.com/Products/FPX/UserGuide/download/Footprint%20Expert%20Surface%20Mount%20Families.pdf

So I think the footprints are correct, although it looks a bit strange. So I'll close this issue for now. But if there is any problem (e.g. with soldering), please let me know.

closed 2023-09-15 by ubruhin