LibrePCB-Libraries/LibrePCB_Base.lplib #127: Single OP

eto-, 2023-10-08:
I'm going to work on TI library adding a set of high-speed op-amp (OPA356, LMH6624, LMH6629 and so on).

The main question is why do you represent the single opamp with an external component for power pins?
It is very non standard and IMHO only confuses the user. Even in your [tutorial](https://librepcb.org/docs/quickstart/create-library-elements/symbol/) you have the usual symbol. 
I can easily add to the TI library but before adding base symbols I would like to understand the logic.

Moreover for many of TI amplifiers there are multiple variants on the package ([OPA356](https://www.ti.com/lit/ds/symlink/opa356.pdf?ts=1696794825933&ref_url=https%253A%252F%252Fwww.google.com%252F) exist in SOT-23-5, SOIC-8 and VSSOP-8): is there a way to implement it?

Finally can you add to the base library the SOT-23-5? It is in your tutorial but not in the library. It is confusing for new users not being able to find components listed in the tutorial.

ubruhin, 2023-10-08:
This not a design decision, there was simply no such combined OpAmp symbol added yet. But it can be added, and the component can be extended with a symbol variant for that so this change would be fully backwards compatible.

> Moreover for many of TI amplifiers there are multiple variants on the package ([OPA356](https://www.ti.com/lit/ds/symlink/opa356.pdf?ts=1696794825933&ref_url=https%253A%252F%252Fwww.google.com%252F) exist in SOT-23-5, SOIC-8 and VSSOP-8): is there a way to implement it?

Sure, you just need to create separate [devices](https://librepcb.org/docs/quickstart/create-library-elements/device/) for them.

> Finally can you add to the base library the SOT-23-5?

It is already there, it's just called differently: SOT95P280X145-5 and you would have found it when typing "sot-23-5" into the package chooser dialog :wink: Note that the term SOT23-5 is not standardized and thus can be ambiguous. Therefore we use IPC7351 naming. The tutorial is just simplified a bit for readability.

eto-, 2023-10-09:
> > Finally can you add to the base library the SOT-23-5?
> 
> It is already there, it's just called differently: SOT95P280X145-5 and you should find it when typing "sot-23-5" into the package chooser dialog 😉 Note that the term SOT-23-5 is not standardized and thus can be ambiguous. Therefore we use IPC7351 naming. The tutorial is just simplified a bit for readability.

Oh then the fact that there is sot23-3 but not sot23-5 with the standard name confused me.

ubruhin, 2023-10-21:
I think all the topics asked in this issue are resolved in the mean time by several PRs, so we can close this.

closed 2023-10-21 by ubruhin