LibrePCB-Libraries/LibrePCB_Base.lplib #134: Missing inductors

eto-, 2023-11-02:
I realized that there are no inductors in the base library.

Can you add the Package categories Chip Inductor and Through-Hole Inductor?
The chip inductors can be probably generated by the generate_chip.py script with some small fixes. I'm not expert enough in the internal details to modify it. The standard chip inductors are 0402,0603 and 0805 (imperial): I've never seen smaller.

ubruhin, 2023-11-03:
Yeah I can take care of this. Currently I'm just not sure whether we need to distinguish between inductors and ferrite beads, both at package level (are packages identical or not?) and on device level (do ferrite beads require a separate component/symbol indicating the ferrite core?). Unfortunately I'm not experienced enough in inductors to know whether every part is clearly either an inductor or a ferrite bead or if they cannot always be distinguished...

eto-, 2023-11-03:
Simple ferrite beads are identical in packages (SMD or THT) to inductors.
The difference is not the presence of the ferrite (most inductors have them in any case) but the losses of the core: for inductors the core must have low losses (to preserve the field energy), for ferrite beads the high losses are wanted (to dissipate noise). 
I guess this distinction is too fine to create parallel structures: when populating the BOM the user will have to know what to use.

ubruhin, 2023-11-03:
Alright, thanks for clarifying :+1: Sounds like we don't need to distinguish them, so I'll create just generic inductor packages + devices.

ubruhin, 2023-11-05:
Here my suggestion for chip inductors: https://github.com/LibrePCB-Libraries/LibrePCB_Base.lplib/pull/135

Any feedback welcome! Otherwise I'll merge in a few days.

eto-, 2023-11-05:
They looks fine: I did not experience in past inductors smaller than 0401 (IM), but having the device is harmless.

Thanks

PR #135 (2023-11-06): Add generic chip inductors
##### SUMMARY

- Add package categories for inductors:
    [image]
- Add packages & devices for generic chip inductors, generated by https://github.com/LibrePCB/librepcb-parts-generator/pull/126
    [image]
    [image]

##### DATASHEETS / REFERENCES

See https://github.com/LibrePCB/librepcb-parts-generator/pull/126.

Closes #134.

##### OPEN QUESTIONS / UNRESOLVED ISSUES

See https://github.com/LibrePCB/librepcb-parts-generator/pull/126.

##### CHECKLIST

- [x] I have read and followed the library conventions¹.
- [x] For packages, I followed IPC7351C (see details in  [... cut]
closed 2023-11-05 by eto-