LibrePCB-Libraries/LibrePCB_Base.lplib #149: Add SMD tantalum capacitors

eto-, 2024-12-02:
I noticed that the library provides packages for SMD polarized capacitors CAPPM* but they are not used for real devices.
These packages miss the 3D model as well.

I guess a script can be updated to automatically generate both the 3D and the devices, but I'm not expert enough to modify it.

These packages are typically used for tantalum capacitors and could be useful in many design.

eto-, 2025-01-23:
I checked the generate_chip.py script but it is not easy to modify it for tantalum capacitors.
I could probably copy the new generate_sod.py script into generate_tantalum.py for having the 3D symbol added.

However since I'm not particularly fluent in python, I'm not sure it will be very nice.

ubruhin, 2025-01-24:
So this is about two topics:
- 3D models for the existing CAPPM packages: This should be relatively easy. In `generate_chip.py` the `generate_3d()` function already has a conditional on package type, so we can add another case to generate different packages for CAPPM. The only question is probably if the 3D models should be yellow (often used for tantalum capacitors) or black (more generic) 🤔 
- Adding devices for these packages. Here I'm not really experienced what devices exist in real world (basically never using tantalium capacitors...). I suspect tantalium capacitors are still a relatively generic thing, worth adding generic devices (e.g. for each of the CAPPM packages, add a device named "Capacitor Tantalium <EIA-CODE>", e.g. "Capacitor Tantalium 3216-10"). Doing so would be relatively easy too with the `generate_chip.py`.

eto-, 2025-01-26:
I worked on the code but the modifications were not minimal.

I created the PR

[image]

[image]

PR #156 (2025-01-31): Add dev & 3D models for tantalum capacitors
##### SUMMARY

Generated by https://github.com/LibrePCB/librepcb-parts-generator/pull/139

- Add 3D models for CAPPM packages
- Add devices for generic tantalum capacitors

##### DATASHEETS / REFERENCES

Closes #149 

##### OPEN QUESTIONS / UNRESOLVED ISSUES

N/A

##### CHECKLIST

- [x] I have read and followed the library conventions¹.
- [x] For packages, I followed IPC7351C (see details in library conventions).
- [x] For modified elements, I bumped their version number².
- [ ] I already used the new library elements in a design and verified their c [... cut]
closed 2025-01-31 by ubruhin